robotology/icub-hri #33: Better repository name

pattacini, 2017-04-20:
@Tobias-Fischer @towardthesea @jypuigbo @clement-moulin-frier @dcam0050 

Any better idea for the name of the repository? Maybe `icub-hri`? What do you think?
In the end, `icub-client` does not seem to be too much telling.

Tobias-Fischer, 2017-04-20:
Hmmmm .. yeah, you're right. At the same time, I don't have a better idea at the moment. I'll think about it :+1:

towardthesea, 2017-04-25:
How about `icub-repertoire` :smile:

Tobias-Fischer, 2017-04-26:
`icub-toolbox`? `icub-hri-toolbox`? That's the best I came up so far with ;)

pattacini, 2017-04-26:
I like `icub-hri-toolbox` actually.

Tobias-Fischer, 2017-05-02:
What do other people think about `icub-hri-toolbox`? @stephane-lallee @towardthesea @clement-moulin-frier @jypuigbo @dcam0050?

clement-moulin-frier, 2017-05-02:
Sounds good to me, as long as it well acknowledged in the `readme` and in the paper where it comes from.
Cheers.

Tobias-Fischer, 2017-05-02:
Ahhh .. well, I guess the title of the paper would then start with "iCub-HRI-Toolbox blablabla", rather than "iCub-Client blablabla" ..

Tobias-Fischer, 2017-05-23:
Hi,
I just spoke to Yiannis, and he likes `icub-hri` (without the toolbox). What do you think? Fine with everyone?
Best, Tobi

pattacini, 2017-05-23:
Isn't `icub-hri` a bit too wide and ambitious? In the sense that we'll cover every possibility for doing HRI with the iCub...

Tobias-Fischer, 2017-05-23:
Yiannis thinks the toolbox makes it sound like a toy :P
What do the other guys think?

pattacini, 2017-05-23:
`icub-hri-client`? 😏

clement-moulin-frier, 2017-05-23:
I kind of agree that icub-hri is very generic, since the iCub is mostly used for HRI all over the world. Some propositions:

- icub-subsystems
- icub-api (at the end what we propose is an API, isn't it?)
- icub-yarp-abstraction
- icub-abstraction
- icub-skills
- icub-primitives
- icub-user-level
- icub-in-da-place
- ok I stop ...

pattacini, 2017-05-24:
@clement-moulin-frier nonetheless `hri` is a good buzzword that should be in 'cause we're doing HRI too; my only concern was that alone is too generic; however, I'm fine with that

Tobias-Fischer, 2017-05-24:
Yiannis still prefers `icub-hri`, but he'd also be fine with `icub-hri-client` as suggested by @pattacini. We still got a couple of days to think about.
While `icub-hri` might be too generic, I find most of your proposals too specific @clement-moulin-frier .. it's more than just subsystems, it's more than just abstraction, ... ;-).

clement-moulin-frier, 2017-05-29:
Paul proposes `icub-h5w`. It might work if we write a paragraph explaining that the functionalities of the library attempts at solving the H5W questions: how (action), what (object recognition), where (tracking) etc. I have some text material for this.
It can also be combined with `hri` and `client`, e.g. `icub-h5w-client`.
What do you think?

pattacini, 2017-05-29:
@clement-moulin-frier that's fine with me too.

Tobias-Fischer, 2017-05-29:
Hmmm .. I can ask Yiannis about it tomorrow. If so, I'd definitely combine it with `hri`.

closed 2017-05-31 by pattacini