robotology/icub-hri #35: Docker image

Tobias-Fischer, 2017-04-26:
Hi,
I had a chat with @dcam0050 and he said that he has a docker image which we could adapt for `icub-client`. I think docker is definitely interesting, so let's put the Dockerfile in this repo. @dcam0050 could you configure it such that the Python bindings (#34) are working as well?

Best,
Tobi

Tobias-Fischer, 2017-05-05:
Hey @dcam0050 - I saw the recent commit with the docker files. Is that all we need? If so, I think we can close this issue - however, could you please write a sentence or two in the paper regarding docker (In "Download, Licensing, Compatibility"?

dcam0050, 2017-05-08:
That's all we need to compile the dockerfile into an image ... people can also go to dockerhub and download it as well from dcamilleri13/icubclient once I upload it ... then it's a matter of preference whether you compile it or download it

Tobias-Fischer, 2017-05-08:
Cool, I'm just trying this right now. I get a few errors though right at the beginning:
`/bin/sh: 1: route: not found` and `/bin/sh: 1: nc: not found`; according to https://github.com/mhamrah/docker-envconsul/issues/1 and https://github.com/mhamrah/docker-envconsul/commit/315d682eb4e45339d14e4212e2776e4649a29b4e you should `apt-get install net-tools netcat` before running those commands. Could you please do that?

Other than that, we can close this once Docker is mentioned in the paper (and possible README) :+1:

dcam0050, 2017-05-16:
Ok so I've completed the above. I am also modifying the image to install karmaWysiwyd as it is not yet installed.

Tobias-Fischer, 2017-05-16:
Thanks @dcam0050. Looks all good to me - except that `apt-get install net-tools netcat` still appears after the first call to `route`, which I guess still throws the errors above.

dcam0050, 2017-05-16:
Have you tried to do a make configure first?

dcam0050, 2017-05-16:
I've pushed an update including those libraries in the make configure recipe. Can you check if it works now please?

Tobias-Fischer, 2017-05-16:
Great, I didn't see the `make configure` step. Should all be working fine now :+1: 
Will close here as soon as the Docker image is uploaded on https://hub.docker.com/r/dcamilleri13/icub-client/
Thanks a lot @dcam0050

dcam0050, 2017-05-16:
cool ... will do that tomorrow once I get to the university because it's a hefty 30GB

Tobias-Fischer, 2017-05-20:
Seems like it's up, closing here :)

closed 2017-05-20 by Tobias-Fischer