robotology/icub-hri #37: Documentation reactiveLayer

Tobias-Fischer, 2017-05-12:
Cleanup is done (#16), documentation remains to be written.

Tobias-Fischer, 2017-05-12:
Reminder to myself: The tutorial on proactive tagging needs updating, it contains wrong references to followingOrder.

Tobias-Fischer, 2017-05-16:
Fixed the tutorial in https://github.com/robotology/icub-client/commit/d18558d9b35dcb922523887af7c8a286c96b88e1 - the rest is up to you @jypuigbo @clement-moulin-frier

Tobias-Fischer, 2017-05-30:
Hi @jypuigbo,
The documentation looks good now, I'm just doing some more cleanup.
A quick question. In `sensationManager`, it seems like `kp_entities` and `k_entities` are not used. Can they be removed? Also, for the other items, it seems like they are just infinitely expanded every time as there is no `u_entities.clear()` etc. anywhere. Should they be local variables within `handleEntities()` instead?
Thanks, Tobi

Tobias-Fischer, 2017-05-30:
We can close this issue once the above is resolved :+1:

jypuigbo, 2017-05-31:
I'm checking that right now.

jypuigbo, 2017-05-31:
I see what you mean. As you prefer. I can just remove every instance of `*_entities` except the ones used, as they were used for WYSIWYD but not necessarily here. Now they might just be noisy for understanding the code. 
Also, I would make them local (as they were originally). If you give me a go signal, I change it now. 

J.

Tobias-Fischer, 2017-05-31:
@jypuigbo Yes please, go ahead and change. Thanks a lot :)

jypuigbo, 2017-05-31:
Sorry, I didn't realize they were used in `get_property`. I just cleared the bottles at the start of `handleEntities` is called, then. 

Let me know if anything else is needed.

Tobias-Fischer, 2017-05-31:
All good, thanks a lot!

closed 2017-05-31 by Tobias-Fischer