robotology/icub-hri #39: opcClient throws seg-fault with different name

towardthesea, 2017-05-18:
I tried to run **iol2opc** in different context, i.e. with pure [IOL application](https://github.com/robotology/iol/blob/master/app/scripts/iol.xml.template). For **OPC** I kept the name *memory* as in xml file, only changed `db` param to `--no-load-db` and `--no-save-db` as in *WYSIWYD* context. I also changed the `opcName` in `iol2opc` 's config file to `memory`. 

Then I ran **iol2opc** module in terminal and got a `seg fault` at [opc initialization](https://github.com/robotology/icub-client/blob/master/src/modules/iol2opc/src/module.cpp#L884).

towardthesea, 2017-05-18:
The terminal output is as following:
```
iol2opc --from config.ini
yarp: Port /iol2opc/world/opc:rpc active at tcp://10.0.0.140:10456/
yarp: Sending output from /iol2opc/world/opc:rpc to /memory/rpc using tcp
Segmentation fault (core dumped)
```

Tobias-Fischer, 2017-05-18:
Hi @towardthesea,
I cannot reproduce the problem locally, it's working fine here.
Can you please provide some debug info:
1. `which iol2opc`
2. `gdb iol2opc`, then type `r`+Enter, then `bt`+Enter, paste the output here.

Thanks :)

towardthesea, 2017-05-18:
Hi @Tobias-Fischer 
The problem disappears when I try it again here :) 
So I close this now. If it happens again, I will collect all output :smile:

closed 2017-05-18 by towardthesea