robotology/icub-hri #41: Seg Fault when running guiUpdater

towardthesea, 2017-05-24:
Hi 
I have just got the `seg fault` when trying to run **guiUpdater** from this repo. When I tried to run the one from WYSIWYD, there is no problem. Following was the terminal output:
```
guiUpdater
||| configuring
[INFO]Display skeleton status: false
yarp: Port /guiUpdater/world/opc:rpc active at tcp://10.0.0.140:10270/
yarp: Sending output from /guiUpdater/world/opc:rpc to /OPC/rpc using tcp
yarp: Port /guiUpdater/gui:o active at tcp://10.0.0.140:10336/
yarp: Port /guiUpdater/guiBase:o active at tcp://10.0.0.140:10337/
yarp: Port /guiUpdater/rpc active at tcp://10.0.0.140:10338/
Segmentation fault (core dumped)
```

Tobias-Fischer, 2017-05-24:
Hi,
Can you please try again after applying the latest commit? If it still doesn't work, could you please add the information obtained after running the module with `gdb` (backtrace)?

Thanks, Tobi

towardthesea, 2017-05-24:
Thanks @Tobias-Fischer , your [fix](https://github.com/robotology/icub-client/commit/2ddd84edbd0f9b10b7451d852a8079dd245788bb) really solves it.

I also help my new module of skeleton, by the way!

closed 2017-05-24 by towardthesea

Tobias-Fischer, 2017-05-24:
What do you mean by new module of skeleton?

Tobias-Fischer, 2017-05-24:
Glad it fixes the problem!

towardthesea, 2017-05-24:
I am developing a module to directly estimate the skeleton from stereo-cameras, which re-uses some libraries from `icub-client`, especially `opcClient`. Hope can introduce it soon (of course our of this repo :smile:)
Thanks again!

Tobias-Fischer, 2017-05-24:
Ahhh I see. That would be great! Let me know if I can help.