robotology/icub-hri #43: iCub.getPartnerName not finding partner

dcam0050, 2017-10-12:
Hey everyone I'm having an issue with getPartnerName and consequently with lookAtPartner. Whenever I try and run the following code I get `[WARNING]No partner present was found!`

I've tracked it down to `icub-client/src/libraries/icubclient/src/clients` line 251 ```list<shared_ptr<Entity>> lEntities = opc->EntitiesCacheCopy();```

The returned list only has one item in it which is icub whereas both icub and partner are present in the OPC when I query it via /OPC/rpc.

I've added a picture here of the modules I'm running in case I'm missing one

[image]

And this is the example code I'm trying

```#!/usr/bin/env python
import yarp
import icubclient
import sys 

yarp.Network.init()
iCub = icubclient.ICubClient("Demo")

rfClient = yarp.ResourceFinder()
rfClient.setVerbose(True)
rfClient.setDefaultContext("icubClient")
rfClient.setDefaultConfigFile("client.ini")
rfClient.configure(sys.argv)

if not iCub.connect():
    print "not connected"
d = iCub.getPartnerName(verbose=True)
print d

iCub.lookAtPartner()
iCub.say("Hello demoing here")
iCub.home("all")
iCub.lookAtPartner()
iCub.say("bye bye")
iCub.home("all")

iCub.close()
yarp.Network.fini()```

Tobias-Fischer, 2017-10-12:
Hey, try whether a `iCub.opc.checkout()` before the `iCub.lookAtPartner()` helps.

Best, Tobi

dcam0050, 2017-10-13:
Thanks Tobi! Worked like a charm

closed 2017-10-13 by dcam0050