robotology/icub-hri #46: Make iol2opc compatible with OpenCV 3.3

Tobias-Fischer, 2017-10-22:
Recent changes in OpenCV3.3 introduce a new API to create new trackers (https://github.com/opencv/opencv_contrib/commit/3ac9e242549a85572739afd232559d2e4baa7e68); see also [3.2 docs](https://docs.opencv.org/3.2.0/d0/d0a/classcv_1_1Tracker.html) and [3.3 docs](https://docs.opencv.org/3.3.0/d0/d0a/classcv_1_1Tracker.html). We should update `iol2opc` to be compatible both with OpenCV >=3.3 and OpenCV 3.0-3.2 (see https://www.learnopencv.com/how-to-find-opencv-version-python-cpp/). Here is a failing build on Travis: https://travis-ci.org/robotology/icub-hri/jobs/290396093

Once this has been updated, let's update Travis and remove the `-DICUBCLIENT_BUILD_IOL2OPC=FALSE` flag. This will enable building `iol2opc` in OSX builds, whereas Linux builds will detect that OpenCV3 is not installed and not build these (a CMake warning will be issued).

pattacini, 2017-10-22:
Hi @Tobias-Fischer 
I'm not a great fan of using #ifdef directives as it would be required as described in https://github.com/robotology/iol/issues/25.

It'd be better to pick a version of OpenCV and stick to it.

Tobias-Fischer, 2017-10-23:
Hey @pattacini,
Hmmmm .. while I fully agree that the OpenCV people messed up (especially seeing that they could have just kept the old way of creating a tracker besides the new way and marking it deprecated ..), I don't think it's a viable solution of forcing people to stick to a particular version of OpenCV. Especially in `icub-hri`, which is meant to be a user friendly repo suitable for beginners as well. If someone uses `brew`, there is no way of specifying the OpenCV version (as far as I can tell), and (s)he might just give up using this repo as the compilation fails.
What do you think? I can write the `#ifdef` stuff if it causes too much pain for you :P

Best, Tobi

pattacini, 2017-10-23:
Ok 😞
I understand the context of `icub-hri` -  which is a bit different than `iol` - and I'll do the update.

pattacini, 2017-10-23:
https://github.com/robotology/icub-hri/commit/3c9e8d3412f109e527fbcd3b2122536bdad84783 fixes this.

closed 2017-10-23 by pattacini