robotology/icub-hri #50: Issue with homeostasis

dcam0050, 2018-01-12:
Hey everyone hope you had a good holiday! 

I've been running into an issue with `homeostasis`. It keeps saying `Waiting for sensations` meaning that the received bottle is empty.

I've tracked down the port  which `homeostasis` is receiving from: `/sensationManager/toHomeo` and put a print statement to verify what is being transmitted and it's transmitting the following:

```[INFO](unknown 4) (known 0) 
[INFO](agent partner) (bodypart unknown_self) (object unknown_object_23) (object unknown_object_24)```

The result is that when the drive on the plot decays below the homeostatic level an action is not being triggered.

I'm using the standard configuration files that come with icub-hri with abm and babbling removed from the subsystem list since those are not present in icub-hri.

Any suggestions of what I can do?

Tobias-Fischer, 2018-01-12:
Hi @dcam0050,
Could you please specify where you put the `print` statement, and how it precisely looks like?
Also, in which configuration file do you need to remove `abm` and `babbling`?
Can you run `homeostasis` in verbose mode, and see whether you get any more output? As far as I can tell, with `verbose=false`, there is never any output other than the mentioned `Waiting for sensations` ..

dcam0050, 2018-01-12:
1) This is where I put the yInfo statements in `opcSensation.cpp`:

``void OpcSensation::publish()
{
    Bottle res = handleEntities();

    if (!Network::isConnected("/opcSensation/toHomeo:o", "/homeostasis/fromSensations:i"))
        Network::connect("/opcSensation/toHomeo:o", "/homeostasis/fromSensations:i");
    yarp::os::Bottle &bot = homeoPort.prepare();
    bot.clear();
    yarp::os::Bottle key;
    key.clear();
    key.addString("unknown");
    key.addInt(res.get(0).asInt());
    bot.addList()=key;
    key.clear();
    key.addString("known");
    key.addInt(res.get(2).asInt());
    bot.addList()=key;
    homeoPort.write();

    yInfo() << bot.toString();

    yarp::os::Bottle &unkn = unknown_entities_port.prepare();
    unkn.clear();
    unkn.append(*res.get(1).asList());
    unknown_entities_port.write();

    yInfo() << unkn.toString();

    yarp::os::Bottle &kn = known_entities_port.prepare();
    kn.clear();
    kn.append(*res.get(3).asList());
    known_entities_port.write();

    yInfo() << kn.toString();
}``

I removed:
 `recognitionOrder` from `icub-client/app/reactiveLayer/behaviorManager/conf/default.ini` because it's not recognised as a behaviour
 `babbling` from `icub-client/app/reactiveLayer/behaviorManager/conf/client.ini`
 'babbling' from `icub-client/app/proactiveTagging/conf/client.ini'
I removed 'abm' from my own files while making them similair to the provided ones

I'll try running homeostasis in verbose mode to see what the output is

dcam0050, 2018-01-12:
Ok so I've attached the log file for homeostasis with verbose set to true
[homeolog.log](https://github.com/robotology/icub-hri/files/1626840/homeolog.log)

and reading from `/opcSensation/toHomeo:o` shows `(unknown 2) (known 0)`

Also I've noticed that in the `drive_control_gui.py` , when I press Update list for Drive Control I get exploration and demonstration which is good. However when I press Update list for Behavior Control I get an empty list.

Tobias-Fischer, 2018-01-12:
For some reason, there are two drives with name `exploration`, which is weird. Also, the values never change. So it's worth having a look what's going on there. I don't know the internals though, maybe @jypuigbo or @clement-moulin-frier can help.

Tobias-Fischer, 2018-01-16:
Any news on this @dcam0050?

dcam0050, 2018-01-16:
I haven't had the time to debug this further yet. I'm waiting to see if @jypuigbo or @clement-moulin-frier have any insights to offer first.

Tobias-Fischer, 2018-03-05:
Hey @dcam0050, did you every figure this out?

Tobias-Fischer, 2018-08-06:
Last try to get an update before closing @dcam0050

dcam0050, 2018-08-07:
I will try this again around beginning of September as I'm busy otherwise
at the moment. Feel free to close the issue and I can reopen it later if
need be.

closed 2018-11-24 by Tobias-Fischer