robotology/icub-hri #52: Issues with iol2opc_IDL

towardthesea, 2018-07-24:
I got the following issues when recompile `icub-hri` with newest `yarp` and `icub-main` (both in devel)
```
make 
Scanning dependencies of target icubclientIDLClients
[  1%] Building CXX object src/libraries/icubclientIDLClients/CMakeFiles/icubclientIDLClients.dir/__/__/idl_generated_code/src/iol2opc_IDL.cpp.o
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp: In member function ‘virtual bool iol2opc_IDL::train_object(const string&)’:
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:249:28: error: cannot declare variable ‘helper’ to be of abstract type ‘iol2opc_IDL_train_object’
   iol2opc_IDL_train_object helper;
                            ^
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:9:7: note:   because the following virtual functions are pure within ‘iol2opc_IDL_train_object’:
 class iol2opc_IDL_train_object : public yarp::os::Portable {
       ^
In file included from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Port.h:14:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/WireLink.h:12,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Wire.h:12,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/include/iol2opc_IDL.h:7,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:4:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
     virtual bool write(ConnectionWriter& writer) const override = 0;
                  ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp: In member function ‘virtual bool iol2opc_IDL::remove_object(const string&)’:
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:259:29: error: cannot declare variable ‘helper’ to be of abstract type ‘iol2opc_IDL_remove_object’
   iol2opc_IDL_remove_object helper;
                             ^
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:18:7: note:   because the following virtual functions are pure within ‘iol2opc_IDL_remove_object’:
 class iol2opc_IDL_remove_object : public yarp::os::Portable {
       ^
In file included from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Port.h:14:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/WireLink.h:12,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Wire.h:12,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/include/iol2opc_IDL.h:7,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:4:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
     virtual bool write(ConnectionWriter& writer) const override = 0;
                  ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp: In member function ‘virtual bool iol2opc_IDL::remove_all()’:
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:269:26: error: cannot declare variable ‘helper’ to be of abstract type ‘iol2opc_IDL_remove_all’
   iol2opc_IDL_remove_all helper;
                          ^
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:27:7: note:   because the following virtual functions are pure within ‘iol2opc_IDL_remove_all’:
 class iol2opc_IDL_remove_all : public yarp::os::Portable {
       ^
In file included from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Port.h:14:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/WireLink.h:12,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Wire.h:12,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/include/iol2opc_IDL.h:7,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:4:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
     virtual bool write(ConnectionWriter& writer) const override = 0;
                  ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp: In member function ‘virtual bool iol2opc_IDL::change_name(const string&, const string&)’:
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:279:27: error: cannot declare variable ‘helper’ to be of abstract type ‘iol2opc_IDL_change_name’
   iol2opc_IDL_change_name helper;
                           ^
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:35:7: note:   because the following virtual functions are pure within ‘iol2opc_IDL_change_name’:
 class iol2opc_IDL_change_name : public yarp::os::Portable {
       ^
In file included from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Port.h:14:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/WireLink.h:12,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Wire.h:12,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/include/iol2opc_IDL.h:7,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:4:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
     virtual bool write(ConnectionWriter& writer) const override = 0;
                  ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp: In member function ‘virtual bool iol2opc_IDL::set_object_persistence(const string&)’:
[... 29 lines cut ...]
                  ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp: In member function ‘virtual void iol2opc_IDL::pause()’:
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:308:21: error: cannot declare variable ‘helper’ to be of abstract type ‘iol2opc_IDL_pause’
   iol2opc_IDL_pause helper;
                     ^
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:62:7: note:   because the following virtual functions are pure within ‘iol2opc_IDL_pause’:
 class iol2opc_IDL_pause : public yarp::os::Portable {
       ^
In file included from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Port.h:14:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/WireLink.h:12,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Wire.h:12,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/include/iol2opc_IDL.h:7,
                 from /usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:4:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
     virtual bool write(ConnectionWriter& writer) const override = 0;
                  ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:34:18: note: 	virtual bool yarp::os::Portable::write(yarp::os::ConnectionWriter&) const
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp: In member function ‘virtual void iol2opc_IDL::resume()’:
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:316:22: error: cannot declare variable ‘helper’ to be of abstract type ‘iol2opc_IDL_resume’
   iol2opc_IDL_resume helper;
                      ^
/usr/local/src/robot/icub-hri/src/idl_generated_code/src/iol2opc_IDL.cpp:69:7: note:   because the following virtual functions are pure within ‘iol2opc_IDL_resume’:
 class iol2opc_IDL_resume : public yarp::os::Portable {
```

pattacini, 2018-07-24:
@towardthesea did you try to regenerate the corresponding IDL files?

pattacini, 2018-07-30:
@towardthesea what's the status of this?

towardthesea, 2018-08-01:
Solved!
Thanks @pattacini !

closed 2018-08-01 by towardthesea

Tobias-Fischer, 2018-08-01:
Hi @towardthesea,
Could you please commit the updates IDLs? See https://travis-ci.org/robotology/icub-hri/jobs/410832233

Thanks, Tobi